theyosh/TerrariumPI #966: [BUG]: fresh install won't work

Snertjong, 2024-10-29:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.12.0
 - Install: [Manual] 
 - Raspberry PI: 4b
 - OS: Latest version and the July version  Raspberry pi os legacy 32 bit 

**Describe the bug**
fresh install TerrariumPI Every installation I do I get an error when starting after installation
tried about 10 times now.
Bought new SD card same problem.

manual start
Job for terrariumpi.service failed because the control process exited with error code.
See "systemctl status terrariumpi.service" and "journalctl -xe" for details.

Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit terrariumpi.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 1.
Oct 29 22:22:35 Terrapi systemd[1]: terrariumpi.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit terrariumpi.service has entered the 'failed' state with result 'exit-code'.
Oct 29 22:22:35 Terrapi systemd[1]: Failed to start TerrariumPI.
░░ Subject: A start job for unit terrariumpi.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit terrariumpi.service has finished with a failure.
░░
░░ The job identifier is 5008 and the job result is failed.
Oct 29 22:22:35 Terrapi systemd[1]: terrariumpi.service: Consumed 2.949s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit terrariumpi.service completed and consumed the indicated resources.

rens@Terrapi:~ $ journalctl -xe
░░ Support: https://www.debian.org/support
░░
░░ The unit terrariumpi.service completed and consumed the indicated resources.
Oct 29 22:22:56 Terrapi systemd[1]: Starting TerrariumPI...
░░ Subject: A start job for unit terrariumpi.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit terrariumpi.service has begun execution.
░░
░░ The job identifier is 5194.
Oct 29 22:22:58 Terrapi python[939]: Traceback (most recent call last):
Oct 29 22:22:58 Terrapi python[939]:   File "/home/rens/TerrariumPI/terrariumPI.py", line 20, in <module>
Oct 29 22:22:58 Terrapi python[939]:     from terrariumEngine import terrariumEngine
Oct 29 22:22:58 Terrapi python[939]:   File "/home/rens/TerrariumPI/terrariumEngine.py", line 2, in <module>
Oct 29 22:22:58 Terrapi python[939]:     from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
Oct 29 22:22:58 Terrapi python[939]:   File "/home/rens/TerrariumPI/hardware/webcam/__init__.py", line 2, in <module>
Oct 29 22:22:58 Terrapi python[939]:     import terrariumLogging
Oct 29 22:22:58 Terrapi python[939]:   File "/home/rens/TerrariumPI/terrariumLogging.py", line 15, in <module>
Oct 29 22:22:58 Terrapi python[939]:     from terrariumNotification import terrariumNotification
Oct 29 22:22:58 Terrapi python[939]:   File "/home/rens/TerrariumPI/terrariumNotification.py", line 36, in <module>
Oct 29 22:22:58 Terrapi python[939]:     from terrariumDatabase import NotificationMessage, NotificationService, Sensor, Relay, Button
Oct 29 22:22:58 Terrapi python[939]:   File "/home/rens/TerrariumPI/terrariumDatabase.py", line 8, in <module>
Oct 29 22:22:58 Terrapi python[939]:     from terrariumUtils import terrariumUtils
Oct 29 22:22:58 Terrapi python[939]:   File "/home/rens/TerrariumPI/terrariumUtils.py", line 23, in <module>
Oct 29 22:22:58 Terrapi python[939]:     from cryptography.fernet import Fernet
Oct 29 22:22:58 Terrapi python[939]: ModuleNotFoundError: No module named 'cryptography'
Oct 29 22:22:59 Terrapi systemd[1]: terrariumpi.service: Main process exited, code=exited, status=1/FAILURE

**To Reproduce / Installation steps**
manual installation steps

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Please add screenshots of the [debug screens](https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/):

**Logfiles**
Please add the log file found in the log folder: `log/terrariumpi.log`. All sensitive information should be stripped, and save to upload.
No logfiles

**Additional context**
Add any other context about the problem here.

theyosh, 2024-10-29:
Found the problem. Do a `git pull` and rerun the installer. It should now installer a lower version of the library cryptography which was causing the problem.

Snertjong, 2024-10-29:
> Found the problem. Do a `git pull` and rerun the installer. It should now installer a lower version of the library cryptography which was causing the problem.

Yes excellent! it's alive! 

I already thought I was doing something wrong..

theyosh, 2024-10-30:
> I already thought I was doing something wrong..
You did NOTHING wrong... some external code was removed. And I had to downgrade a Pyhton library

Snertjong, 2024-10-30:
> > I already thought I was doing something wrong..
> 
> You did NOTHING wrong... some external code was removed. And I had to downgrade a Pyhton library

That's what I found out after trying a hundred times!
I'm going to pick up this project again after years and make new paladarium!

thanks!

closed 2024-10-30 by Snertjong