theyosh/TerrariumPI #967: [BUG]:

MakoShark2, 2024-11-02:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.12.0
 - Install: [Docker]
 - Raspberry PI: 4B
 - OS: Bookworm

**Describe the bug**
Missing translation under: Area configuration > tab variation.

**To Reproduce / Installation steps**
N/A

**Expected behavior**
N/A

**Screenshots**
[image]

**Logfiles**
N/A

**Additional context**
N/A

theyosh, 2024-11-02:
Translations is a community effort: https://github.com/theyosh/TerrariumPI?tab=readme-ov-file#translations

MakoShark2, 2024-11-02:
> Translations is a community effort: https://github.com/theyosh/TerrariumPI?tab=readme-ov-file#translations

I know, but translation is marked as complete (100%) for the language I'm using (Portuguese).

theyosh, 2024-11-02:
Is has not been added to the translation. So that needs some code changes from my side.

But the variation tab is not a stable feature. Also it is not really documented. So not sure if you want to use it...

theyosh, 2024-11-03:
If you want, you can now add new translations: https://weblate.theyosh.nl/translate/terrariumpi/gui/pt/?q=state:%3Ctranslated

But again, it could not be all clear what it should do. It is a complex part of the system which is not being used I guess. Else there would be lot more questions about it.

MakoShark2, 2024-11-03:
OK. Was trying to use the variation but it does seem to be not working as it should. I'll leave it then.

theyosh, 2024-11-03:
Than I close this issue also.

closed 2024-11-03 by theyosh