theyosh/TerrariumPI #968: [BUG]:

kahuwi14, 2024-11-09:
**Setup:**
 - TerrariumPI version: 4.12.0
 - Install: Docker
 - Raspberry PI: 3B
 - OS: Bookworm

After upgrading the OS to Bookworm and stepping up from 4.11. to 4.12.0, I can't start the container again.
I also tried to fix the missing part within the db, but couldn't find something..
Did I miss something?

**Logfiles**
terrariumpi-1  | 2024-11-09 20:57:01,636 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.12.0 on a Raspberry Pi 3 Model B Rev 1.2 ...
terrariumpi-1  | 2024-11-09 20:57:01,701 - INFO    - terrariumEngine       - Loaded 32 settings in 0.06 seconds.
terrariumpi-1  | 2024-11-09 20:57:01,708 - INFO    - terrariumEngine       - Loaded language 'de'.
terrariumpi-1  | 2024-11-09 20:57:01,715 - ERROR   - terrariumWeather      - Loading exception: Weather url 'https://api.openweathermap.org/data/3.0/weather?q=XXXX,de&appid=XXXX' is not valid! Please check your source
terrariumpi-1  | 2024-11-09 20:57:10,105 - INFO    - terrariumEngine       - Start loading total power and water usage
terrariumpi-1  | 2024-11-09 20:57:11,229 - INFO    - terrariumEngine       - Loaded total power and water usage in 1.12 seconds.
terrariumpi-1  | 2024-11-09 20:57:11,231 - INFO    - terrariumEngine       - Loading existing sensors from database.
terrariumpi-1  | 2024-11-09 20:57:11,863 - INFO    - terrariumSensor       - Loaded sensor DHT22 sensor humidity named 'Luftfeuchtigkeit' at address '3' with value 95.30% in 0.61 seconds.
terrariumpi-1  | 2024-11-09 20:57:11,911 - INFO    - terrariumSensor       - Loaded sensor DHT22 sensor temperature named 'Temperatur' at address '3' with value 17.50C in 0.05 seconds.
terrariumpi-1  | 2024-11-09 20:57:11,986 - INFO    - terrariumSensor       - Loaded sensor HC-SR04 ultrasonic ranging sensor distance named 'Wasserstand' at address '7,11' with value 6.45cm in 0.07 seconds.
terrariumpi-1  | 2024-11-09 20:57:11,995 - INFO    - terrariumEngine       - Scanning for new sensors ...
terrariumpi-1  | 2024-11-09 20:57:21,253 - INFO    - terrariumEngine       - Loaded 3 sensors in 10.02 seconds.
terrariumpi-1  | 2024-11-09 20:57:21,255 - INFO    - terrariumEngine       - Loading existing relays from database.
terrariumpi-1  | 2024-11-09 20:57:23,898 - INFO    - terrariumRelay        - Loaded relay Sonoff (Tasmota) relay named 'Taglicht' at address 'XXXX#1' value 0.00 in 2.57 seconds.
terrariumpi-1  | 2024-11-09 20:57:24,138 - INFO    - terrariumRelay        - Loaded relay Sonoff (Tasmota) relay named 'Nachtlicht' at address 'XXXX#2' value 100.00 in 0.24 seconds.
terrariumpi-1  | 2024-11-09 20:57:24,326 - INFO    - terrariumRelay        - Loaded relay Sonoff (Tasmota) relay named 'Beregnung' at address 'XXXX#3' value 0.00 in 0.19 seconds.
terrariumpi-1  | 2024-11-09 20:57:24,511 - INFO    - terrariumRelay        - Loaded relay Sonoff (Tasmota) relay named 'Heizmatte' at address 'XXXX#4' value 0.00 in 0.18 seconds.
terrariumpi-1  | 2024-11-09 20:57:24,514 - INFO    - terrariumEngine       - Scanning for new relays ...
terrariumpi-1  | No GEMBIRD SiS-PM found. Check USB connections, please!
terrariumpi-1  | 2024-11-09 20:57:41,324 - INFO    - terrariumEngine       - Loaded 4 relays in 20.07 seconds.
terrariumpi-1  | 2024-11-09 20:57:41,326 - INFO    - terrariumEngine       - Loading existing buttons from database.
terrariumpi-1  | 2024-11-09 20:57:41,615 - INFO    - terrariumButton       - Loaded Magnetic (door) button named 'Frontscheibe' at address '31' value 1.00 in 0.26 seconds.
terrariumpi-1  | 2024-11-09 20:57:41,617 - INFO    - terrariumEngine       - Loaded 1 buttons in 0.29 seconds.
terrariumpi-1  | 2024-11-09 20:57:41,619 - INFO    - terrariumEngine       - Loading existing webcams from database.
terrariumpi-1  | 2024-11-09 20:57:41,634 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
terrariumpi-1  | 2024-11-09 20:57:41,635 - INFO    - terrariumEngine       - Loading existing enclosures from database.
terrariumpi-1  | Traceback (most recent call last):
terrariumpi-1  |   File "/TerrariumPI/terrariumPI.py", line 23, in <module>
terrariumpi-1  |     terrariumEngine = terrariumEngine(__version__)
terrariumpi-1  |   File "/TerrariumPI/terrariumEngine.py", line 177, in __init__
terrariumpi-1  |     self.__load_existing_enclosures()
terrariumpi-1  |   File "/TerrariumPI/terrariumEngine.py", line 1256, in __load_existing_enclosures
terrariumpi-1  |     list(enclosure.areas),
terrariumpi-1  |   File "/TerrariumPI/terrariumEnclosure.py", line 24, in __init__
terrariumpi-1  |     self.load_areas(areas)
terrariumpi-1  |   File "/TerrariumPI/terrariumEnclosure.py", line 65, in load_areas
terrariumpi-1  |     self.add(terrariumArea(area.id, self, area.type, area.name, area.mode, area_setup))
terrariumpi-1  |   File "/TerrariumPI/terrariumArea.py", line 85, in __init__
terrariumpi-1  |     self.load_setup(setup)
terrariumpi-1  |   File "/TerrariumPI/terrariumArea.py", line 1035, in load_setup
terrariumpi-1  |     super().load_setup(data)
terrariumpi-1  |   File "/TerrariumPI/terrariumArea.py", line 359, in load_setup
terrariumpi-1  |     self._time_table()
terrariumpi-1  |   File "/TerrariumPI/terrariumArea.py", line 194, in _time_table
terrariumpi-1  |     if "weather" == self.mode
terrariumpi-1  | AttributeError: 'NoneType' object has no attribute 'sunrise'

theyosh, 2024-11-09:
It cannot load weather information... Not sure why. You could try go back to a lower version.

What happens when you take the url https://api.openweathermap.org/data/3.0/weather?q=XXXX,de&appid=XXXX and fill in your details. You should be able to open it in your browser.

Because that url is causing somehow an error...

kahuwi14, 2024-11-09:
ahh sorry, I forgot to mansion, that the Problem isn't the weather-Error.
That's because of a faulty URL, but without a GUI, I cannot change it..
But that's not the problem, why the container doesn't start.

theyosh, 2024-11-09:
>  But that's not the problem, why the container doesn't start.

That is the issue!. Because of no weather data, it cannot determine the sunrise time, and it crashes on that. And it will not continue to startup further. And I do take care of the API 3.0 urls.

But I see that either the entered url is wrong, or my logging is wrong. The correct url is:

https://api.openweathermap.org/data/2.5/weather?q=[YOUR_CITY],DE&appid=[YOUR_API_ID]

In the logging it shows 3.0 in the url. That is not correct. That should be still 2.5.

So if the url with the 2.5 does work with your details. I can give you an SQL command to fix it without the GUI. But first I need to know if the weather url is working with 2.5 in the url.

kahuwi14, 2024-11-09:
ahh shit.... i just checked the terrariumArea.py and realised, that it really is the problem.. 

openweather changes the API to 3.0, thats what I did before the update..
And because of caching or whatever, the URL in the Log showed a result, so I updated.... f***

theyosh, 2024-11-09:
Ok, but we can fix that pretty easily. Make sure you have sqlite3 installed:

`sudo apt install sqlite3`

Stop the docker container.

Go to the data folder. I do not know what you have put in your docker-composer.yaml file . But I expect `/opt/terrariumpi/` So enter the data folder: 

`cd /opt/terrariumpi/data`

now open the database with sqlite3: `sqlite3 terrariumpi.db`

You should get a prompt like:

```bash
SQLite version 3.27.2 2019-02-25 16:06:06
Enter ".help" for usage hints.
sqlite> 
```

Now run: `update setting set value = '[YOUR_FULL_WEATHER_URL]' where id = 'weather_source'`

Press CTRL+D to close the database. And now you can start your docker container up.

kahuwi14, 2024-11-09:
used the 2.5-API with the q-attribute and that fixed it.

Thank's a lot!

closed 2024-11-09 by kahuwi14