theyosh/TerrariumPI #970: [FEATURE]: Embeddable Links for Graphics

t00thpick1, 2024-11-26:
**Is your feature request related to a problem? Please describe.**
Writing an html kiosk dashboard for logging+history on feedings/bowel movements/etc for my lizard and wanted to embed some of the temperature/humidity graphics from my TerrariumPi setup directly in my dashboard for a quick glance at terrarium conditions.    Would be nicer to integrate if I didn't have to render the graphs myself with API data but could instead just iframe/embed the graphs as needed.

**Describe the solution you'd like**
Links such as /api/sensors/{id}/embed to render just the graphic for a given sensor such as below.

[image]

theyosh, 2024-11-28:
I am afraid that this is not easy to make. The GUI is made in a reactive javascript framework, so I cannot just pick out the graphs for embedding in a different page.

And it will take a lot of time to make a the graphs embed-able. So I am afraid you have to make you own graphs.
I used https://bernii.github.io/gauge.js/ for the gauge graph, which is pretty old, but does still work and https://www.chartjs.org/ for the graphs.

Sorry, but this is something that I cannot make. There are API calls to get the sensor data and make your own graphs.

t00thpick1, 2024-11-29:
Thats fair, I was hoping it would be easy to pull out just the components but I haven't looked too hard into the svelte code as I'm unfamiliar with svelte.

t00thpick1, 2024-11-29:
If I made some nice re-usable  chart.js scripts for integration would they be welcome somewhere in the codebase as resources for others?

closed 2024-11-29 by t00thpick1