theyosh/TerrariumPI #975: Rugged I2C Moisture Sensor Issue

KDPPD, 2024-12-19:
Been interested in this project for awhile and decided to finally do it but now I'm pretty much stuck on the last sensor.

I purchased this I2C chirp sensor:
https://www.robotshop.com/products/rugged-i2c-soil-moisture-sensor?gad_source=1&gclid=CjwKCAiAgoq7BhBxEiwAVcW0LHC8pwh7UGqfl7dKvULHO3RFbgWzN5LBikgDXBQXCjeQMHVRhxWGIhoCuagQAvD_BwE

I2C moisture sensor is detected on 0x20.
[image]
I2C 0x10 = DockerPI Relay
I2C 0x44 = SHT3XD

I've calibrated utilzing the chirp.py which seems to work on 0x20.
[image]

GUI Settings:
[image]

Log Error:
2024-12-18 17:58:23,928 - ERROR   - hardware.sensor       - Error updating sensor CHIRP sensor moisture named 'Soil Moisture' at address '0x20'. Check your hardware! Error getting new data from sensor CHIRP sensor moisture named 'Soil Moisture' at address '0x20': division by zero

What am I missing here?

Appricate your project and so far it's been a breeze, thanks for all your hardwork.

KDPPD, 2024-12-22:
It works! Excellent job. I was definitely way off from finding that answer.

closed 2024-12-22 by KDPPD

theyosh, 2024-12-22:
Euh, I only fixed the GUI which did not show the min and max moisture calibration values. But that have solved it?

But nice that it works now.