theyosh/TerrariumPI #977: [BUG]: TPI doesn't fully start up after first or second RPI reboot, possibly related to kasa or temp probes

threedom, 2025-01-06:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.12.2 (and 4.7, and 4.9.3)
 - Install: Manual & Docker
 - Raspberry PI: 4
 - OS: Bookworm (and Buster and Bullseye)
 - 6 Kasa Outlets
 - 2 DS18B20 temp probes

**Describe the bug**
Recently, my setup of TerrariumPI (4.7) suddenly stopped working. My 2 1wire probes reported 0F. Once I attempted to restart the system, it wouldn't load the web interface anymore and SSH wouldn't work either.

I have spent the last 3 days trying to get back up and running - unsuccessfully. 
I seem to face a recurring error where something related to my kasa relays fails (according to the logs) and fully bricks the RPI.
I have tried the following changes:

- Stop and start of the docker container
- Reboot and powercycle of the PI
- Full reinstall of TPI with Bullseye, Buster, and Bookworm
- Use of different versions of TPI (4.7, 4.9.3, 4.12.2)
- Docker and Manual install
- New SD card
- Disconnect of all probes and gradual reconnect of probes (and accessories)

With most attempts I can get to a running TPI, log in once or twice, remove / ignore relays (most of my relays are kasa relays that are auto-added on startup, many of which have nothing to do with my terrarium). Then when I perform a reboot via sudo reboot, TPI doesn't reach healthy state and I get the below logs.

Even if I physically disconnect all probes and accessories (e.g. webcam), the error persists and my docker container can't reach healthy state.
And both temp probes provide accurate readings when I manage to connect them. But the system crashes irrespective of that.

I am completely lost at this point. Any help is much appreciated!

**Logfiles**
2025-01-06 07:09:04,096 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.12.2 on a Raspberry Pi 4 Model B Rev 1.4 ...
2025-01-06 07:09:04,131 - INFO    - terrariumEngine       - Loaded 32 settings in 0.03 seconds.
2025-01-06 07:09:04,132 - INFO    - terrariumEngine       - Loaded language 'en_US'.
2025-01-06 07:09:04,858 - INFO    - terrariumEngine       - Start loading total power and water usage
2025-01-06 07:09:05,881 - INFO    - terrariumEngine       - Loaded total power and water usage in 1.02 seconds.
2025-01-06 07:09:05,882 - INFO    - terrariumEngine       - Loading existing sensors from database.
2025-01-06 07:09:05,945 - WARNING - hardware.sensor       - Unable to load sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-01204fba80d8': Unable to load sensor 1wire 1-Wire sensor measuring temperature at address 28-01204fba80d8: Invalid path., retrying in 0.5 seconds...
2025-01-06 07:09:06,454 - WARNING - hardware.sensor       - Unable to load sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-01204fba80d8': Unable to load sensor 1wire 1-Wire sensor measuring temperature at address 28-01204fba80d8: Invalid path., retrying in 0.5 seconds...
2025-01-06 07:09:06,957 - ERROR   - terrariumSensor       - Error loading 1wire temperature named '1-Wire sensor measuring temperature' at address '28-01204fba80d8' with error: Unable to load sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-01204fba80d8': Unable to load sensor 1wire 1-Wire sensor measuring temperature at address 28-01204fba80d8: Invalid path..
2025-01-06 07:09:06,982 - WARNING - hardware.sensor       - Unable to load sensor 1-Wire sensor temperature named 'Yuzu Warm Temp' at address '28-01204fc0d569': Unable to load sensor 1wire Yuzu Warm Temp at address 28-01204fc0d569: Invalid path., retrying in 0.5 seconds...
2025-01-06 07:09:07,485 - WARNING - hardware.sensor       - Unable to load sensor 1-Wire sensor temperature named 'Yuzu Warm Temp' at address '28-01204fc0d569': Unable to load sensor 1wire Yuzu Warm Temp at address 28-01204fc0d569: Invalid path., retrying in 0.5 seconds...
2025-01-06 07:09:07,988 - ERROR   - terrariumSensor       - Error loading 1wire temperature named 'Yuzu Warm Temp' at address '28-01204fc0d569' with error: Unable to load sensor 1-Wire sensor temperature named 'Yuzu Warm Temp' at address '28-01204fc0d569': Unable to load sensor 1wire Yuzu Warm Temp at address 28-01204fc0d569: Invalid path..
2025-01-06 07:09:07,989 - INFO    - terrariumEngine       - Scanning for new sensors ...
2025-01-06 07:09:17,193 - INFO    - terrariumEngine       - Loaded 0 sensors in 11.31 seconds.
2025-01-06 07:09:17,194 - INFO    - terrariumEngine       - Loading existing relays from database.
2025-01-06 07:09:21,635 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '1 Yuzu Heat Panel' at address '192.168.1.251,1' value 0.00 in 4.42 seconds.
2025-01-06 07:09:24,412 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '2 String Lights Tank' at address '192.168.1.251,2' value 100.00 in 2.78 seconds.
2025-01-06 07:09:27,269 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '3 Waterfall' at address '192.168.1.251,3' value 100.00 in 2.86 seconds.
2025-01-06 07:09:29,904 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '4 Growlights Bottom Tank' at address '192.168.1.251,4' value 0.00 in 2.63 seconds.
2025-01-06 07:09:32,713 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '5 Snake Sun' at address '192.168.1.251,5' value 100.00 in 2.81 seconds.
2025-01-06 07:09:32,770 - WARNING - hardware.relay        - Error changing relay Kasa Smart relay named '6 Cooling Fan' at address '192.168.1.251,6' to state 0.0. Error: Unable to query the device 192.168.1.251:9999: [Errno 104] Connection reset by peer, retrying in 0.5 seconds...
2025-01-06 07:09:35,923 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '6 Cooling Fan' at address '192.168.1.251,6' value 0.00 in 3.21 seconds.
2025-01-06 07:09:35,925 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
Traceback (most recent call last):
  File "/opt/venv/lib/python3.11/site-packages/kasa/xortransport.py", line 101, in close
    await writer.wait_closed()
  File "/usr/local/lib/python3.11/asyncio/streams.py", line 364, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 999, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/gevent/_socketcommon.py", line 660, in recv
    return self._sock.recv(*args)
           ^^^^^^^^^^^^^^^^^^^^^^
[... 29 lines cut ...]
  File "/opt/venv/lib/python3.11/site-packages/kasa/smartdevice.py", line 357, in update
    await self._modular_update(req)
  File "/opt/venv/lib/python3.11/site-packages/kasa/smartdevice.py", line 386, in _modular_update
    responses = [
                ^
  File "/opt/venv/lib/python3.11/site-packages/kasa/smartdevice.py", line 387, in <listcomp>
    await self.protocol.query(request) for request in request_list if request
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/kasa/iotprotocol.py", line 43, in query
    return await self._query(request, retry_count)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/kasa/iotprotocol.py", line 64, in _query
    raise ex
  File "/opt/venv/lib/python3.11/site-packages/kasa/iotprotocol.py", line 48, in _query
    return await self._execute_query(request, retry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/kasa/iotprotocol.py", line 86, in _execute_query
    return await self._transport.send(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/venv/lib/python3.11/site-packages/kasa/xortransport.py", line 165, in send
    raise RetryableException(
kasa.exceptions.RetryableException: Unable to query the device 192.168.1.251:9999: [Errno 104] Connection reset by peer

**Additional context**
Add any other context about the problem here.

theyosh, 2025-01-06:
Hmm, to bad that this is happening. I updated the code so it should continue even when a Kasa relays fails. So it should start again in your case.

But there are some things to consider.

1. If you start over with a new SD card, use the OS Bullseye  (Legacy OS). That works the best. Bookworm could work, but is not that well tested. So stick to Bullseye OS. Light image!
2. Could it be that a Kasa relay has changed its IP? So that could case the error about not able to connect to the last relay.
3. That you 1 Wire is missing, could be OS related. On which OS did those errors occur? Again, using Bullseye OS, it should work.
4. This code update, is only available in manual setup. I cannot build a docker image while debugging code issues.

So I suggest, based on the fact that I get the feeling you have a backup of the database, go for a full reinstall with Bullseye OS and the latest code using manual installation.

threedom, 2025-01-06:
Thanks for the reply.
I did a manual install last night using Bullseye that creates the same issues.
However, I figured out that after about an hour, the web interface becomes available again - I am currently running it without any sensors connected to test.
All my relays have fixed IPs from the router's side and haven't changed. And I should emphasize that the odd thing about this one (192.168.1.251:9999) is that after some time, the PI does load it correctly. so it seems sort of arbitrary whether or not it works.

Re/ point 3) - I can try again with my current install but wanted to dig out some other 1-wire sensors I have lying around to see if they cause the same issue or if it could be a corrupted sensor. Since this is a live system that needs to manage my tank I am hesitant to break it again. Might try to get a second pi for redundancy and then experiment on that one to make sure I don't end up doing nightshifts to keep the temps correct.

I'll do another reinstall later today and will post logs to check whether the error persists.

My current version below:

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

theyosh, 2025-01-06:
First concern are the animals. So only do some testing if possible. It is not needed from my part.

OS looks good. 

About the 1-wire sensors, just do what you want. But they should just work in Bullseye OS.

About the relay, it is a bit strange, as there are 6 relays on 1 ip? For Kasa I cannot test, so I cannot optimize it. But normally, this would be optimized by loading all 6 relays at once. And than only 1 connection is needed. And now it is requesting the same device 6 times. This will take some more time. And maybe some side effects like this.

For now, I have only updated the hard loading code so that an Kasa error should not stop loading the system.

threedom, 2025-01-06:
So the 6 relays under 1 IP are a Kasa side thing - it is a single power strip with 6 plugs that can be individually addressed. Since it only creates one connection to the router not sure that there is a different way to address it. [The Power Strip](https://www.amazon.com/Kasa-Smart-Power-Strip-TP-Link/dp/B07G95FFN3/ref=asc_df_B07G95FFN3?mcid=55805b0dc0a43dafb8929792754a4eb3&hvocijid=10623176303531521711-B07G95FFN3-&hvexpln=73&tag=hyprod-20&linkCode=df0&hvadid=692875362841&hvpos=&hvnetw=g&hvrand=10623176303531521711&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9031961&hvtargid=pla-2281435178298&psc=1) for your reference. 

I'll try again with the 1 wires but probably tomorrow. If it fails I want to have time to roll back 😄

threedom, 2025-01-07:
Some additional points:

- I retried the old card and indeed it immediately corrupted when I tried to install TPI and reboot; so new card was necessary
- Did a git pull and upgrade on the running system; still seems to take about an hour for the interface to load but TPI is running when I check the logs. I am still getting the relay error though.
- Ordered a new card so I will experiment a bit more when that arrives later this week (i.e., I'll do a clean install with the changes you implemented). Just don't want to mess with the running system right now. Once the new card is here, I'll test my 1w again. Right now I just want the tank to run for a bit.

threedom, 2025-01-09:
I just did a fresh install and am getting mixed results. Did a startup in debug mode and everything including the web interface loaded in about a minute.
On a normal startup, I am having the same result as before. The web interface loaded after about 20 minutes.

I also noticed that when I follow the logs after startup for longer, I see that eventually another TPI startup is started again and the loop begins once over?

Logs from the run below. It's length but shows when the web interface becomes available.

```
2025-01-08 19:12:42,534 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Waterfall' at address '192.168.1.251,3' value 0.00 in 2.69 seconds.
2025-01-08 19:12:45,207 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Growlights' at address '192.168.1.251,4' value 0.00 in 2.67 seconds.
2025-01-08 19:12:47,927 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Snake Sun' at address '192.168.1.251,5' value 0.00 in 2.72 seconds.
2025-01-08 19:12:47,989 - WARNING - hardware.relay        - Error changing relay Kasa Smart relay named 'Channel Cooling Fan' at address '192.168.1.251,6' to state 0.0. Error: Unable to query the device 192.168.1.251:9999: [Errno 104] Connection reset by peer, retrying in 0.5 seconds...
2025-01-08 19:12:51,072 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Cooling Fan' at address '192.168.1.251,6' value 0.00 in 3.15 seconds.
2025-01-08 19:12:51,313 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Hanging Growlights + Small Tank' at address '192.168.1.67' value 0.00 in 0.24 seconds.
2025-01-08 19:12:51,603 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel String Lights Wall' at address '192.168.1.74,1' value 0.00 in 0.29 seconds.
2025-01-08 19:12:51,791 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel String Lights View' at address '192.168.1.74,2' value 0.00 in 0.19 seconds.
2025-01-08 19:12:52,000 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Stairway' at address '192.168.1.85' value 0.00 in 0.21 seconds.
2025-01-08 19:12:52,001 - INFO    - terrariumEngine       - Scanning for new relays ...
2025-01-08 19:15:43,540 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.12.2 on a Raspberry Pi 4 Model B Rev 1.4 ...
2025-01-08 19:15:43,560 - INFO    - terrariumEngine       - Loaded 32 settings in 0.02 seconds.
2025-01-08 19:15:43,563 - INFO    - terrariumEngine       - Loaded language 'en_US'.
2025-01-08 19:15:44,002 - INFO    - terrariumEngine       - Start loading total power and water usage
2025-01-08 19:15:45,015 - INFO    - terrariumEngine       - Loaded total power and water usage in 1.01 seconds.
2025-01-08 19:15:45,015 - INFO    - terrariumEngine       - Loading existing sensors from database.
2025-01-08 19:15:45,021 - INFO    - terrariumEngine       - Scanning for new sensors ...
2025-01-08 19:15:54,179 - INFO    - terrariumEngine       - Loaded 0 sensors in 9.16 seconds.
2025-01-08 19:15:54,180 - INFO    - terrariumEngine       - Loading existing relays from database.
2025-01-08 19:15:54,957 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Standing Lamp' at address '192.168.1.111' value 0.00 in 0.74 seconds.
2025-01-08 19:15:55,159 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Living Room' at address '192.168.1.140' value 100.00 in 0.20 seconds.
2025-01-08 19:15:55,373 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel String Lights Stairs' at address '192.168.1.196' value 100.00 in 0.21 seconds.
2025-01-08 19:15:55,579 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Video Light' at address '192.168.1.202' value 0.00 in 0.21 seconds.
2025-01-08 19:15:55,813 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Electric Toothbrush' at address '192.168.1.205' value 0.00 in 0.23 seconds.
2025-01-08 19:15:55,988 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Office Lights' at address '192.168.1.222' value 0.00 in 0.17 seconds.
2025-01-08 19:15:56,169 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Bedroom Ceiling' at address '192.168.1.236' value 0.00 in 0.18 seconds.
2025-01-08 19:15:56,369 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Growlight bedroom' at address '192.168.1.241' value 0.00 in 0.20 seconds.
2025-01-08 19:15:58,934 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Heat Panel' at address '192.168.1.251,1' value 0.00 in 2.56 seconds.
2025-01-08 19:16:01,646 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel String Lights Tank' at address '192.168.1.251,2' value 100.00 in 2.71 seconds.
2025-01-08 19:16:04,212 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Waterfall' at address '192.168.1.251,3' value 0.00 in 2.57 seconds.
2025-01-08 19:16:06,752 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Growlights' at address '192.168.1.251,4' value 0.00 in 2.54 seconds.
[... cut ...]
2025-01-08 19:32:25,596 - INFO    - terrariumEngine       - TerrariumPI 4.12.2 is up and running at address: http://0.0.0.0:8090/ in 52.78 seconds.
2025-01-08 19:32:25,597 - INFO    - terrariumEngine       - Starting log tailing.
2025-01-08 19:32:25,608 - INFO    - terrariumEngine       - Starting engine updater with 30.00 seconds interval.
2025-01-08 19:32:25,610 - INFO    - terrariumWebserver    - Running webserver at 0.0.0.0:8090
2025-01-08 19:32:25,861 - INFO    - terrariumEngine       - Starting a new update round with 0 sensors, 18 relays, 0 buttons and 0 webcams.

```

threedom, 2025-01-09:
On another restart (after loading my backup db to this sd card), eventually, TPI was just in a loop of the below. Web interface not loading.
This was after about an hour.

What's unfortunate in this state is that it doesn't run any of the automations. E.g. the heat should be running on a 6 min on, 12 min off timer and it doesn't seem to turn on or off at all.

I'll leave it running over night so see what happens.

```

2025-01-08 20:50:00,324 - INFO    - terrariumEngine       - Loading existing sensors from database.
2025-01-08 20:50:00,332 - INFO    - terrariumEngine       - Scanning for new sensors ...
2025-01-08 20:50:09,475 - INFO    - terrariumEngine       - Loaded 0 sensors in 9.15 seconds.
2025-01-08 20:50:09,476 - INFO    - terrariumEngine       - Loading existing relays from database.
2025-01-08 20:50:12,588 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '1 Yuzu Heat' at address '192.168.1.251,1' value 0.00 in 3.10 seconds.
2025-01-08 20:50:15,230 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '2 String Lights Tank' at address '192.168.1.251,2' value 100.00 in 2.64 seconds.
2025-01-08 20:50:17,775 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '3 Waterfall' at address '192.168.1.251,3' value 0.00 in 2.54 seconds.
2025-01-08 20:50:20,318 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '4 Growlights Bottom Tank' at address '192.168.1.251,4' value 0.00 in 2.54 seconds.
2025-01-08 20:50:22,922 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '5 Yuzu Sun' at address '192.168.1.251,5' value 0.00 in 2.60 seconds.
2025-01-08 20:50:22,986 - WARNING - hardware.relay        - Error changing relay Kasa Smart relay named '6 Yuzu Cooling Fan' at address '192.168.1.251,6' to state 0.0. Error: Unable to query the device 192.168.1.251:9999: [Errno 104] Connection reset by peer, retrying in 0.5 seconds...
2025-01-08 20:50:26,269 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '6 Yuzu Cooling Fan' at address '192.168.1.251,6' value 0.00 in 3.35 seconds.
2025-01-08 20:50:26,272 - INFO    - terrariumEngine       - Scanning for new relays ...
2025-01-08 20:53:08,789 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.12.2 on a Raspberry Pi 4 Model B Rev 1.4 ...
2025-01-08 20:53:08,808 - INFO    - terrariumEngine       - Loaded 32 settings in 0.02 seconds.
2025-01-08 20:53:08,810 - INFO    - terrariumEngine       - Loaded language 'en_US'.
2025-01-08 20:53:09,250 - INFO    - terrariumEngine       - Start loading total power and water usage
2025-01-08 20:53:10,273 - INFO    - terrariumEngine       - Loaded total power and water usage in 1.02 seconds.
2025-01-08 20:53:10,273 - INFO    - terrariumEngine       - Loading existing sensors from database.
2025-01-08 20:53:10,281 - INFO    - terrariumEngine       - Scanning for new sensors ...
2025-01-08 20:53:19,423 - INFO    - terrariumEngine       - Loaded 0 sensors in 9.15 seconds.
2025-01-08 20:53:19,423 - INFO    - terrariumEngine       - Loading existing relays from database.
2025-01-08 20:53:22,559 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '1 Yuzu Heat' at address '192.168.1.251,1' value 0.00 in 3.12 seconds.
2025-01-08 20:53:25,261 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '2 String Lights Tank' at address '192.168.1.251,2' value 100.00 in 2.70 seconds.
2025-01-08 20:53:28,081 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '3 Waterfall' at address '192.168.1.251,3' value 0.00 in 2.82 seconds.
2025-01-08 20:53:30,634 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '4 Growlights Bottom Tank' at address '192.168.1.251,4' value 0.00 in 2.55 seconds.
2025-01-08 20:53:33,594 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '5 Yuzu Sun' at address '192.168.1.251,5' value 0.00 in 2.96 seconds.
2025-01-08 20:53:33,667 - WARNING - hardware.relay        - Error changing relay Kasa Smart relay named '6 Yuzu Cooling Fan' at address '192.168.1.251,6' to state 0.0. Error: Unable to query the device 192.168.1.251:9999: [Errno 104] Connection reset by peer, retrying in 0.5 seconds...
2025-01-08 20:53:36,797 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named '6 Yuzu Cooling Fan' at address '192.168.1.251,6' value 0.00 in 3.20 seconds.
2025-01-08 20:53:36,800 - INFO    - terrariumEngine       - Scanning for new relays ...

```

theyosh, 2025-01-11:
sorry for the late response.

But it looks like that somehow the 6th relay of the kasa device is having issues. I cannot see why, as I do not have a Kasa device. And I think this is strange, as it is the same device as the other 5 relays... so I am also a bit confused...

The problem is that scanning for new relays is taking to much time / is hanging. I am looking into the code to see if I can fix that. Because when you start up TP, it has 3 minutes to fully startup. If that does not happen, TP will be restarted, to try again to startup. As you can see here:

```
2025-01-08 19:25:43,237 - INFO    - terrariumRelay        - Loaded relay Kasa Smart relay named 'Channel Stairway' at address '192.168.1.85' value 0.00 in 0.21 seconds.
2025-01-08 19:25:43,239 - INFO    - terrariumEngine       - Scanning for new relays ...
2025-01-08 19:28:23,487 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.12.2 on a Raspberry Pi 4 Model B Rev 1.4 ...
```

Between the logline about scanning for new relays and the logline where TP startup again, is about 2,5 minute. There it hangs. Probably on the same Kasa device.

But there is a way to start it up. First install the program 'screen': `sudo apt install screen`.
Than run the command `screen` and press enter. This should give you a new shell, where you can start TP in debug mode.
When it starts, you can press `Ctrl+a` and than `d`. This will detach the shell, but TP should continue to run.
Enter `screen -r` to enter the shell again.

https://www.howtogeek.com/662422/how-to-use-linuxs-screen-command/

theyosh, 2025-01-11:
I have updated the code to scan maximal for 30 seconds for new hardware. That should make sure that TP will start at least within the 3 minutes startup window.

But I still do not know why the last kasa relay is causing issues.

theyosh, 2025-01-22:
Any updates on this?

threedom, 2025-01-31:
Hi, sorry for the delayed response.
It's running much more smoothly now. I'm still getting errors form the relays occasionally but most of the time it's just running. So this issue can be closed.

Also for anyone encountering this in the future - faulty 1-wire sensors will mess with the weirdest parts of the system. They corrupted my SD card several times, resulted in the system not loading etc. If you're facing a similar issue to me try disconnecting all 1-wires and then slowly reconnect.

Thank you!

closed 2025-01-31 by threedom