theyosh/TerrariumPI #979: [FEATURE]: Meross Dimmer support MPD100

threedom, 2025-01-31:
Hello, 

I know Meross devices are already generally supported but as far as I can tell the dimmer isn't get and shows up as a standard relay rather than a dimmer in TPi.  Would it be possible to add MPD100 dimmer support? 

https://a.co/d/7T56FuC

This would make day / night dimming and temperature control very easy and cost effective given the low price of the dimmer. 

Thank you!

theyosh, 2025-01-31:
Hi, I guess you have to request that first at: https://github.com/albertogeniola/MerossIot

As the library has no support for that device. Device  type MPD100US is not on the supported list.

theyosh, 2025-02-08:
Closing as I cannot add this device myself.

closed 2025-02-08 by theyosh