theyosh/TerrariumPI #980: [FEATURE]:Request for TAPO p115

jadatklopt, 2025-02-10:
**Is your feature request related to a problem? Please describe.**
When I have the tapo p115 smart plugs inserted in the socket, my terrariumpi doesnt want to start.

I followed the steps from issues #884 and #888 (These issues were about an older model) but cant get it to work.
The provided script in one off the issues will turn the socket on.
But when I start TerrariumPI i get the following error
`File "/home/chris/TerrariumPI/hardware/relay/__init__.py", line 284, in scan_relays
    for relay in relay_device._scan_relays(callback, **kwargs):
  File "/home/chris/TerrariumPI/hardware/relay/kasa_relay.py", line 127, in _scan_relays
    found_devices = __asyncio.run(__scan())
  File "/home/chris/TerrariumPI/terrariumUtils.py", line 65, in run
    return data.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 440, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/home/chris/TerrariumPI/hardware/relay/kasa_relay.py", line 97, in __scan
    await device.update()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/kasa/tapo/tapodevice.py", line 78, in update
    resp = await self.protocol.query("component_nego")
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/kasa/smartprotocol.py", line 63, in query
    return await self._query(request, retry_count)
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/kasa/smartprotocol.py", line 79, in _query
    raise auex
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/kasa/smartprotocol.py", line 68, in _query
    return await self._execute_query(request, retry)
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/kasa/smartprotocol.py", line 166, in _execute_query
    response_data = await self._transport.send(smart_request)
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/kasa/klaptransport.py", line 309, in send
    await self.perform_handshake()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/kasa/klaptransport.py", line 279, in perform_handshake
    local_seed, remote_seed, auth_hash = await self.perform_handshake1()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/kasa/klaptransport.py", line 233, in perform_handshake1
    raise AuthenticationException(msg)
kasa.exceptions.AuthenticationException: Server response doesn't match our challenge on ip 192.168.1.179`

I dont know if it is an feature

theyosh, 2025-02-10:
Hmm TP devices are not support at the moment.... They need a username and password authentication.

I have not added that in my code. I have to look into that.

theyosh, 2025-02-11:
Try running in debug mode which is easier with testing: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

jadatklopt, 2025-02-11:
> So I hope you have installed TP through git (manual). If so, do a `git pull` and restart TP.
> 
> It should now be able to startup, and skip your TAPO relay during startup. If the GUI is available, try adding your relay as a TAPO P100 relay. https://theyosh.github.io/TerrariumPI/hardware/relay/tapo-p100-mini-smart-wi-fi-socket/
> 
> Lets hope this will work.

Yes this is working, it is skipping the kasa_relay and I can start TP with the tapo p115 plugged in.

jadatklopt, 2025-02-11:
Adding the Tapo through the GUI isnt working yet, I get the following error:

2025-02-11 22:14:48,936 - WARNING - hardware.relay        - Could not load hardware for relay TAPO P100/5 relay named 'jan' at address '192.168.1.179': list index out of range, retrying in 0.5 seconds...

theyosh, 2025-02-11:
You entered the address as : `[email@domain.com],[your-password],[ip-relay]` ?

You need to enter your online credentials that you are using https://theyosh.github.io/TerrariumPI/hardware/relay/tapo-p100-mini-smart-wi-fi-socket/#authentication

jadatklopt, 2025-02-11:
My fault, I didnt do that.
Now I get the error:
**Save Error**
Failed to fetch

theyosh, 2025-02-12:
Hmm, than I am not sure if your relay is support with the TAPO relay. If you are handy enough you could do the following:

1. Login with ssh
2. Enable python environment: `cd Terrariumpi` and than `source venv/bin/activate`
3. Create a file called p100.py with the contents below
```python
from PyP100 import PyP100

p100 = PyP100.P100("192.168.X.X", "email@gmail.com", "Password123")  # Creates a P100 plug object

p100.turnOn()  # Turns the connected plug on
p100.turnOff()  # Turns the connected plug off
p100.toggleState()  # Toggles the state of the connected plug
```
Change the IP and login credentials with your values.
4. Run the script with: `python p100.py` and see what happens.

If this is also giving an error, than we need to go with the Kasa route. Which is a bit harder to make.

jadatklopt, 2025-02-12:
> You entered the address as : `[ip-relay],[email@domain.com],[your-password]` ?
> 
> You need to enter your online credentials that you are using https://theyosh.github.io/TerrariumPI/hardware/relay/tapo-p100-mini-smart-wi-fi-socket/#authentication

I think I used the wrong password yesterday, now i get an different error

2025-02-12 19:36:05,803 - ERROR   - PyP100.PyP100         - Failed to initialize protocol AuthProtocol
Traceback (most recent call last):
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 35, in _initialize
    protocol.Initialize()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 110, in Initialize
    raise Exception("Failed to authenticate")
Exception: Failed to authenticate
2025-02-12 19:36:05,929 - ERROR   - PyP100.auth_protocol  - Error: {'error_code': 1003}
2025-02-12 19:36:05,931 - ERROR   - PyP100.PyP100         - Failed to initialize protocol OldProtocol
Traceback (most recent call last):
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 35, in _initialize
    protocol.Initialize()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 244, in Initialize
    result = self._request_raw("handshake", {"key": public_key})
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 173, in _request_raw
    raise Exception(f"Error code: {data['error_code']}")
Exception: Error code: 1003
2025-02-12 19:36:05,935 - WARNING - hardware.relay        - Could not load hardware for relay TAPO P100/5 relay named 'p115' at address '192.168.1.179,redacted@gmail.com,redacted': Failed to initialize protocol, retrying in 0.5 seconds...
2025-02-12 19:36:06,457 - ERROR   - PyP100.PyP100         - Failed to initialize protocol AuthProtocol
Traceback (most recent call last):
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 35, in _initialize
    protocol.Initialize()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 110, in Initialize
    raise Exception("Failed to authenticate")
Exception: Failed to authenticate
2025-02-12 19:36:06,580 - ERROR   - PyP100.auth_protocol  - Error: {'error_code': 1003}
2025-02-12 19:36:06,582 - ERROR   - PyP100.PyP100         - Failed to initialize protocol OldProtocol
Traceback (most recent call last):
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 35, in _initialize
    protocol.Initialize()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 244, in Initialize
    result = self._request_raw("handshake", {"key": public_key})
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 173, in _request_raw
    raise Exception(f"Error code: {data['error_code']}")
Exception: Error code: 1003

jadatklopt, 2025-02-12:
> Hmm, than I am not sure if your relay is support with the TAPO relay. If you are handy enough you could do the following:
> 
> 1. Login with ssh
> 2. Enable python environment: `cd Terrariumpi` and than `source venv/bin/activate`
> 3. Create a file called p100.py with the contents below
> 
> from PyP100 import PyP100
> 
> p100 = PyP100.P100("192.168.X.X", "email@gmail.com", "Password123")  # Creates a P100 plug object
> 
> p100.turnOn()  # Turns the connected plug on
> p100.turnOff()  # Turns the connected plug off
> p100.toggleState()  # Toggles the state of the connected plug
> Change the IP and login credentials with your values. 4. Run the script with: `python p100.py` and see what happens.
> 
> If this is also giving an error, than we need to go with the Kasa route. Which is a bit harder to make.

following error:
python p100.py
Failed to initialize protocol AuthProtocol
Traceback (most recent call last):
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 35, in _initialize
    protocol.Initialize()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 110, in Initialize
    raise Exception("Failed to authenticate")
Exception: Failed to authenticate
Error: {'error_code': 1003}
Failed to initialize protocol OldProtocol
Traceback (most recent call last):
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 35, in _initialize
    protocol.Initialize()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 244, in Initialize
    result = self._request_raw("handshake", {"key": public_key})
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 173, in _request_raw
    raise Exception(f"Error code: {data['error_code']}")
Exception: Error code: 1003
Traceback (most recent call last):
  File "/home/chris/TerrariumPI/p100.py", line 5, in <module>
    p100.turnOn()  # Turns the connected plug on
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 95, in turnOn
    return self.set_status(True)
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 92, in set_status
    return self._set_device_info({"device_on": status})
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 64, in _set_device_info
    return self.request("set_device_info", params)
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 46, in request
    self._initialize()
  File "/home/chris/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 42, in _initialize
    raise Exception("Failed to initialize protocol")
Exception: Failed to initialize protocol

The script you made in in #888 did work

this one:
from PyP100 import PyP100

p100 = PyP100.P100("192.168.X.X", "email@gmail.com", "Password123") #Creates a P100 plug object

p100.handshake() #Creates the cookies required for further methods
p100.login() #Sends credentials to the plug and creates AES Key and IV for further methods

p100.turnOn() #Turns the connected plug on
p100.getDeviceInfo() #Returns dict with all the device info of the connected plug

theyosh, 2025-02-12:
Ok, I see some difference in the scripts. But if the last script, from #888 does work for you, than it should work.
While the documentation says that the `.handshake()` and `.login()` are optional. But apparently we still need them.

But in my TP, I just do that. So I get the feeling that somehow the email address or password was not entered correctly.

Else I have no clue why the test scrip does work, and TP not...

jadatklopt, 2025-02-12:
Working now

theyosh, 2025-02-13:
You mean the script from #888 ? Than it should also work in TP. Make sure you have entered all the information correctly.

`ip,login(email),password`

theyosh, 2025-02-13:
Crap, I just see an update... it is all working!! Nice!

closed 2025-02-14 by theyosh