theyosh/TerrariumPI #985: [BUG]:

blazingrayzer, 2025-03-04:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.1.12
 - Install: [Manual] 
 - Raspberry PI: 4B
 - OS: Buster

**Describe the bug**
Cannot get weather to work, new openweathermap.org account
log file states

2025-03-04 13:59:42,984 - ERROR   - terrariumWeather      - Failed loading weather data: Weather url 'https://api.openweathermap.org/data/2.5/weather?q=Sandton,ZA&appid=********* is not valid! Please check your source

**To Reproduce / Installation steps**
Have tried URL and it appears to work, have tested with solved bug complete URL's with my appid with the same error 

**Additional context**
Unsure if this requires that I need a paid openwwathermap account or not, I am just using the free one from sign up 
have tested other regions, changing URL formats from other blogs and git discussions to no avail

theyosh, 2025-03-04:
Hmm, that is strange. I can use the url with my own appid and that is working:

[image]

What happens when you try to load the url in your browser with your appid? That should produce a page like:

[image]

blazingrayzer, 2025-03-04:
@theyosh - I screwed up and commented out your validation when I was troubleshooting this with the 3.0 api
Cleared that and it started working 

Apologies for wasting your time and thanks for TerrariumPi, has been a really fun project so far.

closed 2025-03-04 by blazingrayzer