theyosh/TerrariumPI #989: Can’t connect to webui

gerben-f1, 2025-03-14:
Hello there,

I have a problem.
Terrarium pi is running but can’t login to webui.

I can connect by putty on windows and says terrariumpi is running.

When i use debug mode i see it starts up but only weather data says error.

line 193, in _time_table
    self.enclosure.weather.sunrise
AttributeError: 'NoneType' object has no attribute 'sunrise'
^CKeyboardInterrupt
2025-03-14T19:01:37Z
Exception ignored in: <module 'threading' from '/usr/lib/python3.9/threading.py'>

theyosh, 2025-03-15:
It looks like that the weather data is not correct. Could you provide more logging? I guess that there is also a warning/error about loading weather data.

gerben-f1, 2025-03-15:
I think its working now.The url was not. Correct i think.M.V.G Gerben

closed 2025-04-06 by theyosh