theyosh/TerrariumPI #990: [Question]: Relay powering everythiong on raspberry boot

kaju666, 2025-03-16:
Hi

Im not sure if it is a bug or misconfiguration but every boot when terrariumpi starts all relays are turning on for split of a second.

My switches are in invert mode. This is classic arduino 4 relay board.

Video of problem:
https://streamable.com/3lmbq4

Any idea how to solve this becouse i can think it is bad for devices such as water pump or heaters.

theyosh, 2025-03-21:
I have attached a zip file with some new code. Could you replace the `hardware/relay/gpio_relay.py` with the attached code?

Make a copy of the original file to your home folder.

And than run in debug mode? Does this work better?

[gpio_relay.py.zip](https://github.com/user-attachments/files/19389492/gpio_relay.py.zip)

theyosh, 2025-04-02:
Did you had time to test the new relay code? As I cannot reproduce your setup, I need some feedback about the code changes. So if you could test it, that would be nice.

theyosh, 2025-04-02:
Did you leave the file in place? And he is still running with the custom relay file? That would indeed mean that the code works.

So I have updated the code in git. Not sure if you do the updates for him, but now you have a changed file which needs to reverted back before you can update again with a new version.

So you could change the relay file back with the command `git checkout hardware/relay/gpio_relay.py` and than a `git pull` to get the latest version/code with this fix in it.

kaju666, 2025-04-09:
Hi

Installed TerrariumPi from scratch on fresh os and this problem still exist.

theyosh, 2025-04-14:
Hmm that is strange, because you should have the new code that was in the zip file... But than I do not know why it happens.

Also when the GPIO relay is loaded, it should not change the state of the GPIO pins. It should stay as is. And after loading the hardware, the relay will be put in last known mode. And than just following the enclosure timer logic.

So I do not know why the relay is toggling during startup.

kaju666, 2025-04-17:
thanks for trying, I'll have to live with it :)

closed 2025-04-17 by kaju666