theyosh/TerrariumPI #994: Can't connect to Tapo P100

Jothoreptile, 2025-04-05:
Hi!
I recently set up TerrariumPi for my _Oecophylla_ and successfully connected one Tapo P100 to it.

[image]

A few days later, I ordered some more Tapo plugs to control my lights, fan, and a bunch of other stuff in the terrarium—because why not! But when I tried to set up the other P100s, I got this error (tested with 2 of the 4 new Tapos):

Save Error
Relay could not be added. Could not load hardware for relay TAPO P100/5 relay named 'Oeco lumière' at address 'XX.XX.XXX': Failed to initialize protocol

Is there any way to fix this "initialize protocol" issue?
I tried removing the Tapo devices from the app, setting them up again, and reconnecting them to my enclosure—but I’m still getting the same error.

theyosh, 2025-04-06:
Hmm, not sure what to do. Is the firmware of the TAPO up to date?
Else you could test with: https://github.com/theyosh/TerrariumPI/issues/888#issuecomment-1911030362

But I do not own this kind of hardware, so I cannot debug it much.

theyosh, 2025-04-07:
This is from the used library: https://github.com/almottier/TapoP100/issues/30

That looks the same error, which looks like wrong combination username/email and password.
Are all plugs available in the app on the phone?

Jothoreptile, 2025-04-07:
Just changed the password (and check on the tapo cloud website), check on the phone got the three plug, retry to add them into TerrariumPi same issues, try to modify the script with the new password and run it. And got this again : 

[image]

theyosh, 2025-04-07:
********* grrr.... fill in the words... that is annoying. 

Just to recap, 1 is working, which is french, the other don't. Do they have the same firmware version? Could it be that the latest firmware of the EU versions is newer, and therefore maybe changed the login code on the plug.

Jothoreptile, 2025-04-07:
Yeah the FR version works perfectly fine but the four UE version doesn't... I really don't understand because the hardware and firmware are the same the only difference is physical (the ground is not the same).
The french one : 
[image]
One of the UE : 
[image]

zavan, 2025-04-19:
I have the same problem.

3 plugs (all from the same box!), 1 works the other 2 throw:

```
Failed to initialize protocol AuthProtocol
Traceback (most recent call last):
  File "/home/zavan/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 35, in _initialize
    protocol.Initialize()
  File "/home/zavan/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 110, in Initialize
    raise Exception("Failed to authenticate")
Exception: Failed to authenticate
Error: {'error_code': 1003}
Failed to initialize protocol OldProtocol
Traceback (most recent call last):
  File "/home/zavan/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 35, in _initialize
    protocol.Initialize()
  File "/home/zavan/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 244, in Initialize
    result = self._request_raw("handshake", {"key": public_key})
  File "/home/zavan/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/auth_protocol.py", line 173, in _request_raw
    raise Exception(f"Error code: {data['error_code']}")
Exception: Error code: 1003
Traceback (most recent call last):
  File "/home/zavan/TerrariumPI/p100.py", line 5, in <module>
    p100.handshake() #Creates the cookies required for further methods
  File "/home/zavan/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 51, in handshake
    self._initialize()
  File "/home/zavan/TerrariumPI/venv/lib/python3.9/site-packages/PyP100/PyP100.py", line 42, in _initialize
    raise Exception("Failed to initialize protocol")
Exception: Failed to initialize protocol
```

They are all using the same account/credentials and they are all on the same firmware: 1.2.5 Build 240411 Rel.143808

Please let me know if I do anything to help debug.

zavan, 2025-04-19:
Based on this, I changed my password to include letter, numbers and symbols and now all 3 are working 🤷🏻

theyosh, 2025-04-22:
@zavan Thanks for finding the fix. @Jothoreptile does this work for you as well?

theyosh, 2025-04-22:
Closing due to fixed

closed 2025-04-22 by theyosh