theyosh/TerrariumPI #1006: [FEATURE]:LumenIZE Support

res2cpu, 2025-05-01:
**Is your feature request related to a problem? Please describe.**
LumenIZE lights are BLE controlled and have support for sunrise/midday/sunset control with UV.
https://arcadialumenize.com/

**Describe the solution you'd like**
If control of these could be integrated into TerriumPI would be a nice option.
I've just bought one myself, the big sway for me to pull the trigger is that they can be controlled via ESPHome/Home Assistant(esp32 BLE) so my plan was to use a GPIO on the RaspPI connected to the ESP32 to trigger ESPHome to turn on the light/UV and control ramp on the ESP32.
If built in it would lower the requirements for people to also use these.

**Additional context**
Here is the link to the ESPHome implementation with the protocols listed under "ble_client" 
https://www.jefrim.nl/posts/integration-arcadia-lumenize-with-esphome

theyosh, 2025-05-04:
Hmm, ik kan zo geen python libraries vinden die deze hardware ondersteunt. Dan is het voor mij lastig om ondersteuning te maken in TP.

Op basis van je blogpost kan ik geen ondersteuning maken. Het lijkt niet heel ingewikkeld, maar ik kan het niet testen. Als jij een python script kan maken die via bluetooth de lampen kan aansturen, dan kan ik kijken of ik het kan integreren in TP.

theyosh, 2025-06-07:
Closing due no response

closed 2025-06-07 by theyosh