theyosh/TerrariumPI #1007: [BUG]: Relay delay is not taken into account

HTPProXy, 2025-05-09:
**Setup:**
 - TerrariumPI version: 4.12.2
 - Install: Docker
 - Raspberry PI: 2B
 - OS: Bullseye

**Describe the bug**
I would like to simulate sunrise and sunset with two non-dimmable lights by switching them on with a delay.
One lamp is to be switched on at sunrise and the second 30 minutes later. At sunset, the procedure should be reversed.

So my idea was to artificially shorten the day by 30 minutes (Hours shift: -0.5) so that a light is switched off 30 minutes before sunset.

I realized the remaining times via relay delays:

1st light:
	- Switch-on delay 30 minutes (=actual sunrise)
	- Switch-off delay 30 minutes (=actual sunset)
2nd light:
	- Switch-on delay 60 minutes (=actual sunrise + 30 minutes)
	- Switch-off delay 0 minutes (=actual sunrise - 30 minutes)

When switching from night to day, this works correctly, the 1st light comes on at sunrise and the 2nd light 30 minutes later.
When changing from day to night, however, both lights are switched off at the same time. The lights are therefore switched off 30 minutes before the actual sunset.

The log file shows that both relays are switched off with a delay of 0 minutes when switching from day to night.
There is actually only one place in the source code where the delay of a relay is overwritten:

```
if (relay.ON if action else relay.OFF) != relay.state and self.state[part]["powered"] == action:
            delay = 0

```

Is it possible that self.state[part]["powered"] has already adopted the new state before the relay is switched?

**To Reproduce / Installation steps**
Setup:
Hours shift: -0.5
1st light:
	- Switch-on delay 30 minutes
	- Switch-off delay 30 minutes
2nd light:
	- Switch-on delay 60 minutes
	- Switch-off delay 0 minutes

https://prnt.sc/E6tqZ5GDu3g2
https://prnt.sc/uBdeMDIAjMt4

**Expected behavior**
When the day changes to night (30mins before the real sunset) the relay with 30 mins switch-off delay should switch off after 30 mins.

**Logfiles**
Logfile is attached.

[terrariumpi_logfile.txt](https://github.com/user-attachments/files/20126791/terrariumpi_logfile.txt)

theyosh, 2025-05-10:
Hmm, I think you found a bug. I have re-created your setup and I am now testing to see if I can find the problem.

Your setup and logic is correct. So what you try to accomplish should be possible.

theyosh, 2025-05-12:
The two lines you mentioned are indeed the problem. But you are running with docker, which is a bit hard to update to testing.

So I will create a new release soon, which should include this fix.

HTPProXy, 2025-05-13:
Thank you, I will update the container as soon as the new image is ready!

theyosh, 2025-06-07:
Sorry for the new release delay. Working on it.