theyosh/TerrariumPI #1009: [BUG]: How to use variation?

HTPProXy, 2025-05-14:
**Setup:**
 - TerrariumPI version: 4.12.2
 - Install: Docker
 - Raspberry PI: 2
 - OS: Bullseye

**Describe the bug**
It is not clear how to use the variation tab. When setting source to "At time" the period/duration field only allows integer. The difference field has a description, that positive and negative values are supported and a number is treated as an absolut value, but every positive value ist replaced to an absolute value after saving. Is it supposed to only support int values?

What exactly does "At time" mean? When this time is reached the value gets changed to the absolute value? Or is the value used until this time is reached?

Nevertheless it does not seem to work for me.

**To Reproduce / Installation steps**
Set variation to "At time" and define some absolute values. In my case:
At time 09 -> 17
At time 11 -> 23
At time 15 -> 27
At time 18 -> 23
At time 19 -> 17

It is 5:10pm atm and the area shows 16-18°C

**Expected behavior**
If "At time" means that the value is set at this time, then the area should show 26-28°C, if it means until this time then it should be 22-24°C.

**Screenshots**
https://prnt.sc/jdFUWyJZOqig
https://prnt.sc/Yv4t8oziANS5

**Logfiles**
The complete logfile can be found here: https://limewire.com/d/o5UOV#9TwxL5TwQk

**Additional context**
Add any other context about the problem here.

theyosh, 2025-05-16:
This variation tab is rather complex to explain and to setup. It was a wish from somebody years ago.

But the idea is that you can set small periods during a 24 hours day/night cycle. In those periods you can increase and decrease sensor values. So `at time` is an absolute time like 13:04 and `after` is in hours. 

So than you can say start at 08:00
After 30 minutes increase sensors by 5
After 60 minutes increase sensors by 5 once more
At 16:00 decrease sensors by 5
After 30 min decrease sensors by 5 once more

Also it should be able to mimic history call weather data from the day before. But again, it is rather complex, and not fully tested. And I think nobody is using it. I am even thinking of disabling that part. Also, I have not coded that part for a long time, so fixing bugs is hard and time consuming.

So I would advise not to use it. That makes it all much easier.

HTPProXy, 2025-05-16:
I found an old screenshot from this section where the at time values are defined using 04:00 AM syntax. Now this field only allows integer values and using them won't work. So maybe the only issue is that the field value gets parsed to integer. I tried to find the piece of code that tries to parse the input but couldn't find it.

theyosh, 2025-05-29:
Based on your code I made the following change
https://github.com/theyosh/TerrariumPI/commit/714ef83a52dda9c73e1d7b9250d7491b46224172

this should than fix it? I cannot really test it, as it takes a specialized setup.
If you can check if this works correctly, that would be nice.

And thanks for the code!

theyosh, 2025-05-29:
I have created development docker images with your code in it. https://hub.docker.com/repository/docker/theyosh/terrariumpi/tags?name=development-&page=1

Than you could test if it works correctly for you.

HTPProXy, 2025-05-29:
Thank you, I will install it this weekend and give you feedback.

HTPProXy, 2025-06-01:
I just did a fresh install of the develeopment-bullseye image, but starting the container fails with:

`Traceback (most recent call last):
  File "/TerrariumPI/terrariumPI.py", line 20, in <module>
    from terrariumEngine import terrariumEngine
  File "/TerrariumPI/terrariumEngine.py", line 2, in <module>
    from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
  File "/TerrariumPI/hardware/webcam/__init__.py", line 23, in <module>
    import cv2
  File "/opt/venv/lib/python3.9/site-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/opt/venv/lib/python3.9/site-packages/cv2/__init__.py", line 153, in bootstrap
    native_module = importlib.import_module("cv2")
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libwebpdemux.so.2: cannot open shared object file: No such file or directory`

It is an old rpi2 but this was my test machine and the stable worked fine.

theyosh, 2025-06-07:
Nice! A new release is underway with this fix.

closed 2025-06-11 by HTPProXy