theyosh/TerrariumPI #1014: [BUG]: Not able to change Calibration fields on Script dimmer (after initial creation)

Guod3, 2025-06-19:
**Setup:**
 - TerrariumPI version: 4.12.3
 - Install:  [Docker]
 - Raspberry PI: [ 4]
 - OS: [Bookworm]

**Describe the bug**
I have a script dimmer operational. When I (try to) change one of the Calibration fields, the red box says:   
"Not all required fields are entered correctly.
'Dimmer frequency in Hz'"

**To Reproduce / Installation steps**
Delete the script dimmer, and create a new one. Enter the fields including calibration, enter 100 or some other amount in "Max power in%". The dimmer functions at the power level specified for the timer schedule. Later try to change the power level (or any other Calibration value) and we get the red box says:   
"Not all required fields are entered correctly.
'Dimmer frequency in Hz'"  

**Expected behavior**
Obviously anything to do with frequency is bogus because this type of relay has not got it.
I would expect the parameters of Calibration to take effect on the relay output as they do if applied when the relay is created.

Guod3, 2025-06-26:
Every time I need to change a field in a Script dimmer Calibration field, I get this error. I can delete and recreate the relay with the desired value change, but this hurts my brain.

Guod3, 2025-06-26:
> I have to look into it. But why do you need to change the relay so often?
>
> Also, you can just fill in the required field with any value, as it will
> not be used in the scripted dimmer. Or do I miss something here??
>
> I hope to have some time soon to look into it.

A script dimmer may use the max power in% field to determine the level that
is ramped up to, for example and it is really good to have ui where you can
change things, rather than editing a script. I'm tending to change things a
lot as I configure my terrarium system with all kinds of sensors and
actuators, and scripts.
BTW,Thanks for the project!

theyosh, 2025-06-29:
I had some problems creating the development docker images. This issue should now be fixed.

Pick the right development version to test https://hub.docker.com/repository/docker/theyosh/terrariumpi/tags?name=development&page=1

Guod3, 2025-06-30:
From the link I get a prompt to login at dockerhub, and when I did it says: "You are not authorized to view this page. See theyosh's profile page instead"

Guod3, 2025-07-21:
> Did you had time to test this?

Sorry, yes I did and it seems fine now. TY