theyosh/TerrariumPI #1015: [BUG]: Setting a relay value randomly fails

HTPProXy, 2025-06-27:
**Setup:**
 - TerrariumPI version: 4.1.12
 - Install: Docker
 - Raspberry PI: 4
 - OS: Bullseye

**Describe the bug**
I am receiving mails that setting the value of an Energie USB relay to a specific value failed with an unknown reason and that there will be another try in 0.5s. The relay gets toggled, but I just don't know, if this is the result of the first or second try. It also seems strange that when failing to set the value to 100, it gets set back to 0 and then to 100 again.
This happens randomly with all 4 ports of the Energenie USB.

**To Reproduce / Installation steps**
Add an Energenie USB relay and change the relay value.

**Expected behavior**
Setting the relay should not end with an error.

**Logfiles**
Logfile is attached.

[terrariumpi_logfile.txt](https://github.com/user-attachments/files/20940684/terrariumpi_logfile.txt)

theyosh, 2025-06-30:
Hmm, based on you logging I see some strange things.

Normally there should be a log line starting with `Toggle the relays for area ...` before changing the relays. I do not see them in the log.
So is the relay changed with an app, or outside TP? 

It should try 3 times to change/read the relay state. But I get the feeling it may not work (well) anymore. Not sure if that is Python version related.

But based on your logfile, it does toggle.
`2025-06-27 06:19:39,476 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' with new value 0.00 in 0.07 seconds.`
`2025-06-27 06:20:09,484 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' with new value 100.00 in 0.05 seconds.`
It did change from 0 to 100. But I do not see any `Toggle...` log line.

But TP should always try to put the relays in the wanted state. So I check every 30 seconds all the relay, and calculate the wanted stated. If it is not correct, TP will try to switch the relays. It is not event based.
This should make it more fail safe when a toggle action is failing once.

HTPProXy, 2025-07-01:
Ok, I think I know why there is no "Toggle" entry in the log.

To simulate sun rise and sun set I created 3 different areas for light:

Light #1: from sun rise to sun set
Light #2: from sun rise + 30 mins to sun set - 30 mins
Light #3: from sun rise + 60 mins to sun set - 60 mins (the relay that hat the issue)

To realize that I use the time offset field (-1h) and add a toogle on delay with 120 mins.

The Toggle entry in the log is written when the area becomes active, the log I posted is the log after 120 mins.

`2025-06-27 04:19:39,238 - INFO    - terrariumArea         - Toggle the relays for area Lighting named 'Bright Sun 50W' part day to state on.
2025-06-27 04:19:39,239 - INFO    - terrariumArea         - Set the relay Beleuchtung, Bright Sun 50W to 100.0 with a delay of 120.00 minutes`
...
`2025-06-27 06:19:39,279 - WARNING - hardware.relay        - Error changing relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' to state 100.0. Error: Error changing relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' to state 100.0. Error: unknown, retrying in 0.5 seconds...
2025-06-27 06:19:39,305 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Arcadia Dragon 54W, hinten' at address '2,01:01:4d:a6:32' with new value 100.00 in 0.08 seconds.
2025-06-27 06:19:39,354 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor humidity named 'Decke' at address '0x44,23' with new value 50.85% in 0.03 seconds.
2025-06-27 06:19:39,468 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor temperature named 'Decke' at address '0x44,23' with new value 29.56C in 0.01 seconds.
2025-06-27 06:19:39,476 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' with new value 0.00 in 0.07 seconds.
2025-06-27 06:19:39,614 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor humidity named 'Umgebung' at address '0x44,24' with new value 51.77% in 0.04 seconds.
2025-06-27 06:19:39,636 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Bright Sun 70W' at address '4,01:01:4d:a6:32' with new value 0.00 in 0.06 seconds.
2025-06-27 06:19:39,724 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor temperature named 'Umgebung' at address '0x44,24' with new value 27.83C in 0.01 seconds.
2025-06-27 06:19:39,853 - INFO    - terrariumArea         - Updated area Lighting named 'Bright Sun 50W' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-06-27 06:19:39,854 - INFO    - terrariumArea         - Updated area Lighting named 'Arcadia Dragon 54W, hinten' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-06-27 06:19:39,854 - INFO    - terrariumArea         - Updated area Lighting named 'Arcadia Dragon 54W, vorne' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-06-27 06:19:39,859 - INFO    - terrariumArea         - Updated area Heating named 'Bright Sun 70W' in 'sensors' mode at enclosure Bartagame in 0.00 seconds.
2025-06-27 06:19:39,862 - INFO    - terrariumEnclosure    - Updated Enclosure Bartagame with 4 areas in 0.01 seconds.
2025-06-27 06:19:39,957 - INFO    - terrariumEngine       - Message of the Day generated in 0.09 seconds
2025-06-27 06:19:39,993 - INFO    - terrariumEngine       - Engine update done in 0.95 seconds. Waiting for 29.05 seconds for the next round.
2025-06-27 06:19:41,446 - INFO    - terrariumArea         - Updated area Lighting named 'Bright Sun 50W' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-06-27 06:19:41,447 - INFO    - terrariumArea         - Updated area Lighting named 'Arcadia Dragon 54W, hinten' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-06-27 06:19:41,447 - INFO    - terrariumArea         - Updated area Lighting named 'Arcadia Dragon 54W, vorne' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-06-27 06:19:41,457 - INFO    - terrariumArea         - Updated area Heating named 'Bright Sun 70W' in 'sensors' mode at enclosure Bartagame in 0.01 seconds.
2025-06-27 06:19:41,464 - INFO    - terrariumEnclosure    - Updated Enclosure Bartagame with 4 areas in 0.02 seconds.
2025-06-27 06:20:09,048 - INFO    - terrariumEngine       - Starting a new update round with 6 sensors, 4 relays, 0 buttons and 0 webcams.
2025-06-27 06:20:09,148 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor temperature named 'Boden' at address '0x44,22' with new value 29.11C in 0.08 seconds.
2025-06-27 06:20:09,166 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Arcadia Dragon 54W, vorne' at address '1,01:01:4d:a6:32' with new value 100.00 in 0.08 seconds.
2025-06-27 06:20:09,258 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor humidity named 'Boden' at address '0x44,22' with new value 49.35% in 0.01 seconds.
2025-06-27 06:20:09,325 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Arcadia Dragon 54W, hinten' at address '2,01:01:4d:a6:32' with new value 100.00 in 0.05 seconds.
2025-06-27 06:20:09,393 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor humidity named 'Decke' at address '0x44,23' with new value 50.79% in 0.03 seconds.
2025-06-27 06:20:09,484 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' with new value 100.00 in 0.05 seconds.`

HTPProXy, 2025-07-03:
There is no need to do a huge rewrite of your codebase to get the retry working. In my case the relay switches, maybe within the next loop but it works.

The reason I set a 120 min delay is that this relay is controlling a UV light. It is not very realistic if the terrarium would get full UV light as soon as the sun rises and stops UV light when the sun sets. In reality the sun starts emitting UV with sunrise and will slowly increase. And it would decrease before sun set. So i decided it would be the best to wait until sun rise + 60 mins to start the UV light and stop it 60 mins before sun set.

To achieve that I adjusted the time offset with -1h (=stop heating 60 mins bevore sun set) and used a delay of 120 mins. So it will start heating 60 mins after sun rise.

theyosh, 2025-07-04:
Hmm, I am looking into this, and it does seems to work for me:

```
2025-07-04 20:08:00,069 - INFO    - terrariumRelay        - Updated relay Energenie LAN relay named 'Dripper plant' at address 'http://*********@192.168.5.150#1' with new value 100.00 in 0.08 seconds.
2025-07-04 20:08:10,650 - WARNING - hardware.relay        - Error getting new data from relay Energenie LAN relay named 'Mister' at address 'http://*********@192.168.5.150#2': Timed out after 10 seconds, retrying in 0.5 seconds...
2025-07-04 20:08:11,251 - INFO    - terrariumRelay        - Updated relay Energenie LAN relay named 'Mister' at address 'http://*********@192.168.5.150#2' with new value 0.00 in 11.07 seconds.
```
Here there is an error that it took more than 10 seconds. And just after that, the update succeeds with 11 seconds of processing. This would suggest that the retry is working, as coded.

So for updating, it seems to work.

For toggling, I am now testing with a bit more logging, in order to see what it does. But I get the feeling it does work, but I do not log it when a toggle action is actual done.

```
2025-07-04 20:08:18,906 - INFO    - terrariumArea         - Set the relay Energenie LAN relay named 'Mister' at address 'http://*********@192.168.5.150#2' to 100.0 with 0 seconds delay
2025-07-04 20:08:19,715 - INFO    - hardware.relay        - Toggled relay Energenie LAN relay named 'Mister' at address 'http://*********@192.168.5.150#2' from state '0.0' to state '100.0' in 0.81 seconds.
2025-07-04 20:08:30,870 - INFO    - terrariumRelay        - Updated relay Energenie LAN relay named 'Dripper plant' at address 'http://*********@192.168.5.150#1' with new value 100.00 in 0.84 seconds.
2025-07-04 20:08:31,061 - INFO    - terrariumRelay        - Updated relay Energenie LAN relay named 'Mister' at address 'http://*********@192.168.5.150#2' with new value 100.00 in 0.08 seconds.
2025-07-04 20:08:42,235 - INFO    - terrariumArea         - Set the relay Energenie LAN relay named 'Mister' at address 'http://*********@192.168.5.150#2' to 0.0 with 0 seconds delay
2025-07-04 20:08:43,131 - INFO    - hardware.relay        - Toggled relay Energenie LAN relay named 'Mister' at address 'http://*********@192.168.5.150#2' from state '100.0' to state '0.0' in 0.90 seconds.
````

Last logline is new. It will ad more logging when using dimmers.... I am now testing it.

theyosh, 2025-07-07:
Made another new development image with the extra relay logging. If you could test this image, and see if there is a new log line after the error in the morning. It should now produce an extra log line every time the relay is changing value.

HTPProXy, 2025-07-19:
After 2 weeks I already thought the issue wouldn't happen anymore, but this morning it finally happended. Here the relevant logs:

`2025-07-19 04:37:52,345 - INFO    - terrariumEngine       - Starting a new update round with 6 sensors, 4 relays, 0 buttons and 0 webcams.
2025-07-19 04:37:52,447 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor temperature named 'Boden' at address '0x44,22' with new value 27.62C in 0.08 seconds.
2025-07-19 04:37:52,471 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Arcadia Dragon 54W, vorne' at address '1,01:01:4d:a6:32' with new value 0.00 in 0.08 seconds.
2025-07-19 04:37:52,558 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor humidity named 'Boden' at address '0x44,22' with new value 41.05% in 0.01 seconds.
2025-07-19 04:37:52,629 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Arcadia Dragon 54W, hinten' at address '2,01:01:4d:a6:32' with new value 0.00 in 0.05 seconds.
2025-07-19 04:37:52,693 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor humidity named 'Decke' at address '0x44,23' with new value 42.42% in 0.03 seconds.
2025-07-19 04:37:52,784 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' with new value 0.00 in 0.05 seconds.
2025-07-19 04:37:52,803 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor temperature named 'Decke' at address '0x44,23' with new value 26.47C in 0.01 seconds.
2025-07-19 04:37:52,940 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Bright Sun 70W' at address '4,01:01:4d:a6:32' with new value 0.00 in 0.05 seconds.
2025-07-19 04:37:52,951 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor humidity named 'Umgebung' at address '0x44,24' with new value 44.83% in 0.04 seconds.
2025-07-19 04:37:53,075 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor temperature named 'Umgebung' at address '0x44,24' with new value 26.87C in 0.01 seconds.
2025-07-19 04:37:53,200 - INFO    - terrariumArea         - Toggle the relays for area Lighting named 'Bright Sun 50W' part day to state on.
2025-07-19 04:37:53,201 - INFO    - terrariumArea         - Set the relay Beleuchtung, Bright Sun 50W to 100.0 with a delay of 120.00 minutes
2025-07-19 04:37:53,202 - INFO    - terrariumArea         - Updated area Lighting named 'Bright Sun 50W' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-07-19 04:37:53,202 - INFO    - terrariumArea         - Updated area Lighting named 'Arcadia Dragon 54W, hinten' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-07-19 04:37:53,203 - INFO    - terrariumArea         - Updated area Lighting named 'Arcadia Dragon 54W, vorne' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-07-19 04:37:53,208 - INFO    - terrariumArea         - Updated area Heating named 'Bright Sun 70W' in 'sensors' mode at enclosure Bartagame in 0.00 seconds.
2025-07-19 04:37:53,211 - INFO    - terrariumEnclosure    - Updated Enclosure Bartagame with 4 areas in 0.01 seconds.
2025-07-19 04:37:53,292 - INFO    - terrariumEngine       - Message of the Day generated in 0.08 seconds
2025-07-19 04:37:53,327 - INFO    - terrariumEngine       - Engine update done in 0.98 seconds. Waiting for 29.02 seconds for the next round.`

2 hours later ;)

`2025-07-19 06:37:53,016 - INFO    - terrariumEngine       - Starting a new update round with 6 sensors, 4 relays, 0 buttons and 0 webcams.
2025-07-19 06:37:53,076 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor temperature named 'Boden' at address '0x44,22' with new value 27.80C in 0.05 seconds.
2025-07-19 06:37:53,099 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Arcadia Dragon 54W, vorne' at address '1,01:01:4d:a6:32' with new value 100.00 in 0.06 seconds.
2025-07-19 06:37:53,187 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor humidity named 'Boden' at address '0x44,22' with new value 41.46% in 0.01 seconds.
2025-07-19 06:37:53,250 - WARNING - terrariumRelay        - Error changing relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' to state 100.0. Error: Error changing relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' to state 100.0. Error: unknown, retrying in 0.5 seconds...
2025-07-19 06:37:53,276 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Arcadia Dragon 54W, hinten' at address '2,01:01:4d:a6:32' with new value 100.00 in 0.07 seconds.
2025-07-19 06:37:53,321 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor humidity named 'Decke' at address '0x44,23' with new value 41.84% in 0.03 seconds.
2025-07-19 06:37:53,437 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT3X sensor temperature named 'Decke' at address '0x44,23' with new value 28.15C in 0.01 seconds.
2025-07-19 06:37:53,444 - INFO    - terrariumRelay        - Updated relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' with new value 0.00 in 0.07 seconds.
[... cut ...]
2025-07-19 06:37:56,987 - INFO    - terrariumArea         - Updated area Lighting named 'Arcadia Dragon 54W, vorne' in 'weather' mode at enclosure Bartagame in 0.00 seconds.
2025-07-19 06:37:56,996 - INFO    - terrariumArea         - Updated area Heating named 'Bright Sun 70W' in 'sensors' mode at enclosure Bartagame in 0.01 seconds.
2025-07-19 06:37:56,999 - INFO    - terrariumEnclosure    - Updated Enclosure Bartagame with 4 areas in 0.02 seconds.right Sun 70W' at address '4,01:01:4d:a6:32' with new value 0.00 in 0.06 seconds.`

theyosh, 2025-07-19:
Based on you log, I am almost sure that the retry works :) Look at the following loglines:

```
2025-07-19 06:37:53,250 - WARNING - terrariumRelay        - Error changing relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' to state 100.0. Error: Error changing relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' to state 100.0. Error: unknown, retrying in 0.5 seconds...
```
and
```
2025-07-19 06:37:56,963 - INFO    - terrariumRelay        - Toggled relay Energenie USB relay named 'Beleuchtung, Bright Sun 50W' at address '3,01:01:4d:a6:32' from state '0.0' to state '100.0' in 3.75 seconds.
```
The last log line is the successfull toggle action. That took 3,75 seconds. And the time between the error and the ok is about 3.7 seconds.

So based on this, I would say, the retry logic is working.

But as a lot of stuff is running in threats, it could see that is not working, or that states are not in the expected state. But I cannot stop other processes when there is an error in another process. So the log can look a bit confusing, but looks good.

So I say, it is all working correctly.