theyosh/TerrariumPI #1018: [BUG]: Unable to get weather data from  openweather  using API ver 3.0 and existing subscribtion

Zubrania, 2025-07-26:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.12.3
 - Install: Docker
 - Raspberry PI:  4B
 - OS:  Bookworm

**Describe the bug**
Unable to get weather data from openweathermap.org  using API ver 3.0  with the following source URL: https://api.openweathermap.org/data/3.0/onecall?lat=52.23&lon=21.01&appid=xxxxxx with  created and activated account @ openwaethermap. with the 

**To Reproduce / Installation steps**
providing the following Source  url  in weather settings https://api.openweathermap.org/data/3.0/onecall?lat=52.23&lon=21.01&appid=xxxxxx

appid have been replaced 

the following output can be seen from  the logs inside container

2025-07-26 12:34:30,277 - ERROR   - terrariumWeather      - Failed loading weather data: Weather url 'https://api.openweathermap.org/data/3.0/onecall?lat=52.23&lon=21.01&appid=********* is not valid! Please check your source

Please provide the steps to reproduce or how you have installed the software.

**Expected behavioк*
curl --location 'https://api.openweathermap.org/data/3.0/onecall?lat=52.23&lon=21.01&appid=xxxxxx'

provides all the data 
{"lat":52.23,"lon":21.01,"timezone":"Europe/Warsaw","timezone_offset":7200,"current":{"dt":1753523345,"sunrise":1753498059,"sunset":1753555022,"temp":300.34,"feels_like":300.95,"pressure":1012,"humidity":53,"dew_point":289.94,"uvi":6.34,"clouds":20,"visibility":10000,"wind_speed":2.06,"wind_deg":180,"weather":[{"id":801,"main":"Clouds","description":"few clouds","icon":"02d"}]},"minutely":[{"dt":1753523400,"precipitation":0},{"dt":1753523460,"precipitation":0},{"dt":1753523520,"precipitation":0},{"dt":1753523580,"precipitation":0},{"dt":1753523640,"precipitation":0},{"dt":1753523700,"precipitation":0},{"dt":1753523760,"precipitation":0},{"dt":1753523820,"precipitation":0},{"dt":1753523880,"precipitation":0},{"dt":1753523940,"precipitation":0},{"dt":1753524000,"precipitation":0},{"dt":1753524060,"precipitation":0},{"dt":1753524120,"precipitation":0},{"dt":1753524180,"precipitation":0},{"dt":1753524240,"precipitation":0},{"dt":1753524300,"precipitation":0},{"dt":1753524360,"precipitation":0},{"dt":1753524420,"precipitation":0},{"dt":1753524480,"precipitation":0},{"dt":1753524540,"precipitation":0},{"dt":1753524600,"precipitation":0},{"dt":1753524660,"precipitation":0},{"dt":1753524720,"precipitation":0},{"dt":1753524780,"precipitation":0},{"dt":1753524840,"precipitation":0},{"dt":1753524900,"precipitation":0},{"dt":1753524960,"precipitation":0},{"dt":1753525020,"precipitation":0},{"dt":1753525080,"precipitation":0},{"dt":1753525140,"precipitation":0},{"dt":1753525200,"precipitation":0},{"dt":1753525260,"precipitation":0},{"dt":1753525320,"precipitation":0},{"dt":1753525380,"precipitation":0},{"dt":1753525440,"precipitation":0},{"dt":1753525500,"precipitation":0},{"dt":1753525560,"precipitation":0},{"dt":1753525620,"precipitation":0},{"dt":1753525680,"precipitation":0},{"dt":1753525740,"precipitation":0},{"dt":1753525800,"precipitation":0},{"dt":1753525860,"precipitation":0},{"dt":1753525920,"precipitation":0},{"dt":1753525980,"precipitation":0},{"dt":1753526040,"precipitation":0},{"dt":1753526100,"precipitation":0},{"dt":1753526160,"precipitation":0},{"dt":1753526220,"precipitation":0},{"dt":1753526280,"precipitation":0},{"dt":1753526340,"precipitation":0},{"dt":1753526400,"precipitation":0},{"dt":1753526460,"precipitation":0},{"dt":1753526520,"precipitation":0},{"dt":1753526580,"precipitation":0},{"dt":1753526640,"precipitation":0},{"dt":1753526700,"precipitation":0},{"dt":1753526760,"precipitation":0},{"dt":1753526820,"precipitation":0},{"dt":1753526880,"precipitation":0},{"dt":1753526940,"precipitation":0}],"hourly":[{"dt":1753520400,"temp":300.18,"feels_like":300.85,"pressure":1012,"humidity":54,"dew_point":290.08,"uvi":5.63,"clouds":28,"visibility":10000,"wind_speed":2.37,"wind_deg":251,"wind_gust":2.62,"weather":[{"id":802,"main":"Clouds","description":"scattered clouds","icon":"03d"}],"pop":0},{"dt":1753524000,"temp":300.34,"feels_like":300.95,"pressure":1012,"humidity":53,"dew_point":289.94,"uvi":6.34,"clouds":20,"visibility":10000,"wind_speed":2.28,"wind_deg":255,"wind_gust":2.35,"weather":[{"id":801,"main":"Clouds","description":"few clouds","icon":"02d"}],"pop":0},{"dt":1753527600,"temp":300.44,"feels_like":300.92,"pressure":1012,"humidity":51,"dew_point":289.42,"uvi":6.38,"clouds":26,"visibility":10000,"wind_speed":2.6,"wind_deg":252,"wind_gust":2.64,"weather":[{"id":802,"main":"Clouds","description":"scattered clouds","icon":"03d"}],"pop":0},{"dt":1753531200,"temp":300.8,"feels_like":301.06,"pressure":1012,"humidity":48,"dew_point":288.81,"uvi":5.49,"clouds":34,"visibility":10000,"wind_speed":3.13,"wind_deg":252,"wind_gust":2.92,"weather":[{"id":802,"main":"Clouds","description":"scattered clouds","icon":"03d"}],"pop":0},{"dt":1753534800,"temp":301.39,"feels_like":301.42,"pressure":1011,"humidity":45,"dew_point":288.34,"uvi":2.64,"clouds":68,"visibility":10000,"wind_speed":3.66,"wind_deg":256,"wind_gust":3.23,"weather":[{"id":803,"main":"Clouds","description":"broken clouds","icon":"04d"}],"pop":0},{"dt":1753538400,"temp":300.88,"feels_like":300.65,"pressure":1011,"humidity":41,"dew_point":286.44,"uvi":1.44,"clouds":84,"visibility":10000,"wind_speed":3.25,"wind_deg":261,"wind_gust":3.13,"weather":[{"id":803,"main":"Clouds","description":"broken clouds","icon":"04d"}],"pop":0},{"dt":1753542000,"temp":299.64,"feels_like":299.64,"pressure":1011,"humidity":41,"dew_point":286.29,"uvi":1.2,"clouds":98,"visibility":10000,"wind_speed":3.4,"wind_deg":300,"wind_gust":2.86,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04d"}],"pop":0},{"dt":1753545600,"temp":298.58,"feels_like":298.35,"pressure":1011,"humidity":45,"dew_point":287.25,"uvi":0.72,"clouds":98,"visibility":10000,"wind_speed":3.4,"wind_deg":316,"wind_gust":3.18,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04d"}],"pop":0},{"dt":1753549200,"temp":298.7,"feels_like":298.56,"pressure":1010,"humidity":48,"dew_point":287.53,"uvi":0.26,"clouds":99,"visibility":10000,"wind_speed":2.29,"wind_deg":344,"wind_gust":2.57,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04d"}],"pop":0},{"dt":1753552800,"temp":297.94,"feels_like":297.86,"pressure":1010,"humidity":53,"dew_point":288.04,"uvi":0,"clouds":99,"visibility":10000,"wind_speed":1.62,"wind_deg":24,"wind_gust":2.5,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04d"}],"pop":0},{"dt":1753556400,"temp":297.3,"feels_like":297.29,"pressure":1011,"humidity":58,"dew_point":288.38,"uvi":0,"clouds":100,"visibility":10000,"wind_speed":1.25,"wind_deg":52,"wind_gust":1.3,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04n"}],"pop":0},{"dt":1753560000,"temp":296.01,"feels_like":295.97,"pressure":1011,"humidity":62,"dew_point":288.62,"uvi":0,"clouds":100,"visibility":10000,"wind_speed":1.29,"wind_deg":106,"wind_gust":1.2,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04n"}],"pop":0},{"dt":1753563600,"temp":295.67,"feels_like":295.65,"pressure":1011,"humidity":64,"dew_point":288.63,"uvi":0,"clouds":100,"visibility":10000,"wind_speed":1.36,"wind_deg":68,"wind_gust":1.54,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04n"}],"pop":0},{"dt":1753567200,"temp":294.93,"feels_like":294.89,"pressure":1011,"humidity":66,"dew_point":288.36,"uvi":0,"clouds":98,"visibility":10000,"wind_speed":1.51,"wind_deg":51,"wind_gust":1.58,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04n"}],"pop":0},{"dt":1753570800,"temp":294.35,"feels_like":294.25,"pressure":1011,"humidity":66,"dew_point":287.99,"uvi":0,"clouds":98,"visibility":10000,"wind_speed":1.16,"wind_deg":37,"wind_gust":1.13,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04n"}],"pop":0},{"dt":1753574400,"temp":290.5,"feels_like":290.01,"pressure":1011,"humidity":66,"dew_point":287.77,"uvi":0,"clouds":98,"visibility":10000,"wind_speed":1.46,"wind_deg":26,"wind_gust":1.68,"weather":[{"id":804,"main":"Clouds","description":"overcast clouds","icon":"04n"}],"pop":0},{ [... cut ...] ouds":100,"pop":1,"rain":11.55,"uvi":7}]}

**Screenshots**
Please add screenshots of the [debug screens](https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/):

**Logfiles**

[terrariumpi_logfile.txt](https://github.com/user-attachments/files/21443055/terrariumpi_logfile.txt)

Please add the log file found in the log folder: `log/terrariumpi.log`. All sensitive information should be stripped, and save to upload.

**Additional context**
Add any other context about the problem here.

theyosh, 2025-07-26:
You are using incorrect weather url: https://theyosh.github.io/TerrariumPI/setup/#setup

Zubrania, 2025-07-26:
Indeed it worked. Thanks.  The reference to API 3.0  in the doc made me  to go wrong way :(

theyosh, 2025-07-27:
@SyM3N I made an update, if you are using `manual` (git) installation, you can do a `git pull` and restart TP.

Than you should also be able to enter the weather url with a (location) id.