theyosh/TerrariumPI #1019: [FEATURE]: Forced ventillation of terrarium using PWM  fans  with fan hubs

Zubrania, 2025-07-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
Add the ability  to manage fan speed or multiple fan speed  with FAN hubs  to react on  humidity/ temperature  changes  and react accordingly  using forced ventilation 

**Additional context**
Add any other context or screenshots about the feature request here.

DZJ8, 2025-07-30:
I am talking about  devices like that [https://noctua.at/en/nv-fh2](url) which can connect multiple 4 pin fans which can control speed and get actual RPM and  be able to control fans, its speed to  provide "forced ventilation" in terrarium  or just cool it down quicker than "natural ventilation"

G4KCM, 2025-08-01:
There are two pwm channels that can be assigned to GPIO, I used them in a BBQ controller.