theyosh/TerrariumPI #1023: [FEATURE]: 8-HUB FTDI USB-Relay support

CypH3r-LE, 2025-09-04:
Hi,

Since we expanded our terrariums, I had to extend the relays. I then installed an 8-channel FTDI USB relay. Now the problem is that relays 5–8 toggle (audible clicking) with every sensor update. Relays 1–4 were recognized without any issues. Relays 5–8 could be added manually and work without problems.

I would like to see support for 8-channel USB relay modules as well.

We use 
- [this Module](https://www.amazon.de/dp/B076CLT2WK?ref=ppx_yo2ov_dt_b_fed_asin_title)
- RasPi 4 Mod. B
- TerrariumPi V. 4.12.5

Auto-Detect: 2,A10MQ963
Manually-Added: 8,A10MQ963

Many thanks in advance.
And thank you very much for all the work you have put into the project so far. We have been using TerrariumPi for two years and are more than thrilled. Excellent work!
(Pictures of the finished setup will follow.)

CypH3r-LE, 2025-09-05:
Thanks for updating the documentation.

The actual feature request, however, referred to the fact that the relays 5-8 click every minute and whether you have any idea why that might be. I suspected the autodiscover function might be the cause (at least from my perspective).

theyosh, 2025-09-15:
The auto-discovery only runs during startup. Or when you manually click on 'scan relays' or 'scan sensors'. So that is not the issue.

But I have also some clicking sounds with the 4 relay board. But this happens 1 or 2 times a day. But it does not toggle the relay. So it looks like that reading out the device can sometimes give a click sound.

I have no clue why this is happening. I do know when I change the code to run all after each other, there is more clicking sounds. Now the code has been '[monkey patched](https://www.gevent.org/api/gevent.monkey.html)' to get more power out of the PI by running more processes parallel. It could be that this has something to do with it.

So for me I hear sometimes the clicking, but there is nothing actually toggled. So I leave it as is. I also do not know how to fix this.

But does the relays toggle at your setup? I hope not, else there is an issue, which will be hard to fix, I am afraid.